LowPowerLab/SPIFlash #4: SPI.transfer(0) returns 255 with not IC present

charlespax, 2015-01-08:
I am using the SPIFlash library on my device and it works great when the flash chip is present. I have found, however, that if the chip is not present, the library hangs when calling `flash.initialize()` from my Arduino sketch.

I have found that when a chip is not present on my hardware `SPI.transfer(0)` returns 255, which makes `readStatus()` always return 255, which makes `busy()` always return 1, which makes `while(busy());` in `command()` loop forever, which makes my firmware hang.

The problem is that a flash chip not present looks like a busy flash chip. I think it would be great if a no present flash chip were not busy.

https://github.com/PaxInstruments/t400-firmware/issues/125

charlespax, 2015-01-08:
I have confirmed the same issue on my hardware with the example sketch compiled in Arduino 1.0.6, `flash.initialize()` hangs forever.

LowPowerLab, 2015-01-09:
The issue is with static/noise on the open drain MISO input. That causes a hang in busy() when non 0 statuses are returned. You should add a 10k pulldown or similar weak pulldown on MISO if it's not used to avoid such open drain misreads caused by static or noise in the circuit. This is safe even with the chip present. A time limit check cannot be safely added in the library because some flash chips can take many seconds to finish executing erases or other multi-block or entire chip operations.

closed 2015-01-09 by LowPowerLab

charlespax, 2015-01-09:
Awesome! Thanks for the tip. I'll make the PCB change.

LowPowerLab, 2015-01-09:
You can test the initialize() and readStatus() functions with this code:

#include <SPIFlash.h>  //get it here: https://github.com/LowPowerLab/SPIFlash
#include <SPI.h>
SPIFlash flash(8, 0xEF30);  // windbond 4mbit flash chip
void setup()
{
  Serial.begin(115200);
}

void loop()
{
  Serial.println("\n******************************************");

  //this should return 0 if flash chip is not present
  //otherwise it indicates a high static environment causing noise to be returned on MIS, in turn causing hangs in initialize
  //
  Serial.print("FLASH readStatus() = "); Serial.println(flash.readStatus());

  //this should return false 
  boolean initializedOK = flash.initialize();
  if (initializedOK)
    Serial.println("INIT PASS, chip present");
  else
    Serial.println("INIT FAIL, chip not present or chip does not match given signature");
  
  delay(2000);
}

charlespax, 2015-01-09:
Using a slightly modified version of your code I tested the hardware.

With a flash chip present the serial output is
```
******************************************
FLASH readStatus() = 0
INIT PASS, chip present
```
With the flash disconnected the serial output is
```
******************************************
FLASH readStatus() = 255
```

Test code:
```
#include <SPI.h>
#include <SPIFlash.h>    // For SPI flash https://github.com/LowPowerLab/SPIFlash )

SPIFlash flash(11, 0xEF40); // windbond 4mbit flash chip
void setup()
{
Serial.begin(115200);
delay(3000);
}

void loop()
{
Serial.println("\n******************************************");

//this should return 0 if flash chip is not present
//otherwise it indicates a high static environment causing noise to be returned on MISO (which is open drain), in turn causing hangs in initialize
//
Serial.print("FLASH readStatus() = "); Serial.println(flash.readStatus());

//this should return false 
boolean initializedOK = flash.initialize();
if (initializedOK)
Serial.println("INIT PASS, chip present");
else
Serial.println("INIT FAIL, chip not present or chip does not match given signature");

delay(2000);
}
```