LowPowerLab/SPIFlash #8: Arduino 1.6

kalanda, 2015-02-21:
This library works perfectly fine on Arduino 1.0.6 but inside the new release 1.6.0 the sketch hangs very soon. Is easy to check using the SPIFlash_ReadWrite example. I think this issue is because of changes on SPI.h library but didn't found any solution.

tbowmo, 2015-03-17:
I see the exact same thing here, after upgrade to arduino 1.6.0, SPIFlash_ReadWrite is crashing..

kalanda, 2015-03-17:
Maybe this fork's commit has solved it but I don't see any pull request to this repo:
https://github.com/TomWS1/SPIFlash/commit/8c32a312ea52d703bafbe2856222f4847491a513

vicatcu, 2015-03-17:
fwiw, seems like this could be Moteino specific, my fork for WildFire seems to still be working fine in arduino 1.6.0

tbowmo, 2015-03-17:
@kalanda tried the one from @TomWS1 repo, it can initialize the SPI flash library successfully

kalanda, 2015-03-20:
There is a pull request https://github.com/LowPowerLab/SPIFlash/pull/9 for @LowPowerLab from @TomWS1 waiting for revision.

closed 2015-07-25 by kalanda