LowPowerLab/SPIFlash #17: class SPIFlash miss members

wenhuizhang, 2016-11-17:
Arduino: 1.6.11 (Mac OS X), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/WenhuiZhang/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/WenhuiZhang/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/WenhuiZhang/Documents/Arduino/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -vid-pid=0X2341_0X0042 -ide-version=10611 -build-path /var/folders/qd/xz6y3zgn0sb8g920b3p_6tph0000gn/T/build437cbc79e0e0f2bf32157c84dc737f09.tmp -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/Users/WenhuiZhang/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino6 -prefs=runtime.tools.avr-gcc.path=/Users/WenhuiZhang/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2 -verbose /Users/WenhuiZhang/Downloads/SPIFlash/examples/SPIFlash_ReadWrite/SPIFlash_ReadWrite.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/WenhuiZhang/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/WenhuiZhang/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/WenhuiZhang/Documents/Arduino/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -vid-pid=0X2341_0X0042 -ide-version=10611 -build-path /var/folders/qd/xz6y3zgn0sb8g920b3p_6tph0000gn/T/build437cbc79e0e0f2bf32157c84dc737f09.tmp -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/Users/WenhuiZhang/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino6 -prefs=runtime.tools.avr-gcc.path=/Users/WenhuiZhang/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2 -verbose /Users/WenhuiZhang/Downloads/SPIFlash/examples/SPIFlash_ReadWrite/SPIFlash_ReadWrite.ino
Using board 'mega' from platform in folder: /Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14
Using core 'arduino' from platform in folder: /Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14
Build options changed, rebuilding all
Detecting libraries used...
"/Users/WenhuiZhang/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/cores/arduino" "-I/Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/variants/mega" "/var/folders/qd/xz6y3zgn0sb8g920b3p_6tph0000gn/T/build437cbc79e0e0f2bf32157c84dc737f09.tmp/sketch/SPIFlash_ReadWrite.ino.cpp" -o "/dev/null"
"/Users/WenhuiZhang/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/cores/arduino" "-I/Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/variants/mega" "-I/Users/WenhuiZhang/Documents/Arduino/libraries/SPIFlash" "/var/folders/qd/xz6y3zgn0sb8g920b3p_6tph0000gn/T/build437cbc79e0e0f2bf32157c84dc737f09.tmp/sketch/SPIFlash_ReadWrite.ino.cpp" -o "/dev/null"
"/Users/WenhuiZhang/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/cores/arduino" "-I/Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/variants/mega" "-I/Users/WenhuiZhang/Documents/Arduino/libraries/SPIFlash" "-I/Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/libraries/SPI/src" "/var/folders/qd/xz6y3zgn0sb8g920b3p_6tph0000gn/T/build437cbc79e0e0f2bf32157c84dc737f09.tmp/sketch/SPIFlash_ReadWrite.ino.cpp" -o "/dev/null"
"/Users/WenhuiZhang/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/cores/arduino" "-I/Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/variants/mega" "-I/Users/WenhuiZhang/Documents/Arduino/libraries/SPIFlash" "-I/Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/libraries/SPI/src" "/Users/WenhuiZhang/Documents/Arduino/libraries/SPIFlash/DMASPI.cpp" -o "/dev/null"
"/Users/WenhuiZhang/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/cores/arduino" "-I/Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/variants/mega" "-I/Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/libraries/SPI/src" "-I/Users/WenhuiZhang/Documents/Arduino/libraries/SPIFlash" "/Users/WenhuiZhang/Documents/Arduino/libraries/SPIFlash/SPIFlash.cpp" -o "/dev/null"
"/Users/WenhuiZhang/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/cores/arduino" "-I/Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/variants/mega" "-I/Users/WenhuiZhang/Documents/Arduino/libraries/SPIFlash" "-I/Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/libraries/SPI/src" "/Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/libraries/SPI/src/SPI.cpp" -o "/dev/null"
"/Users/WenhuiZhang/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/cores/arduino" "-I/Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/variants/mega" "-I/Users/WenhuiZhang/Documents/Arduino/libraries/SPIFlash" "-I/Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/libraries/SPI/src" "/var/folders/qd/xz6y3zgn0sb8g920b3p_6tph0000gn/T/build437cbc79e0e0f2bf32157c84dc737f09.tmp/sketch/SPIFlash_ReadWrite.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/Users/WenhuiZhang/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/cores/arduino" "-I/Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/variants/me [... cut ...]
                            ^
Multiple libraries were found for "SPIFlash.h"
 Used: /Users/WenhuiZhang/Documents/Arduino/libraries/SPIFlash
 Not used: /Users/WenhuiZhang/Documents/Arduino/libraries/SPIFlash-master
Using library SPIFlash at version 2.5.0 in folder: /Users/WenhuiZhang/Documents/Arduino/libraries/SPIFlash 
Using library SPI at version 1.0 in folder: /Users/WenhuiZhang/Library/Arduino15/packages/arduino/hardware/avr/1.6.14/libraries/SPI 
exit status 1
'class SPIFlash' has no member named 'initialize'

eroniki, 2017-02-06:
I was having the same problem. I then realized that there is a naming conflict between Arduino libraries. To be more specific, there is another [SPIFlash](https://github.com/Marzogh/SPIFlash) library which seems to confuse the Arduino library manager. If you update your libraries with the Arduino Library, it is highly likely that you unintentionally overwrite this library with the one above. Thus, I would suggest you have a clear installation of SPIFlash library and stop updating your libraries through Arduino Library Manager.

 I hope it will solve your problem. 

Bests,
M

LowPowerLab, 2017-02-06:
Thanks, looks like [Marzogh](https://github.com/Marzogh/SPIFlash) did not bother to look around when he named his new library, so special thanks goes to him for creating this mess. Not sure there's anything I can do about it. I think people can always add their library through the library manager.

wenhuizhang, 2017-02-15:
@LowPowerLab @eroniki , I tried using AVRASP V2.0, with baud rate of 11520, and it works now :)

closed 2017-02-15 by wenhuizhang