LowPowerLab/SPIFlash #19: Library name

2lostkiwis, 2017-09-06:
Any chance of changing the library name to prevent a conflict with the following:
https://github.com/Marzogh/SPIFlash

Always seems to give a "library update is available" message which is annoying. When you go in to manage the libraries, it comes up with a different SPIFlash by Prajwal Bhattaram - I have never used this library but it somehow seems integrated to the Arduino environment.

[image]

LowPowerLab, 2017-09-08:
Thanks, I am aware of this. I think it would be fair for the other person to rename his since he created it years after I created mine. His is specifically windbond so maybe he could rename to "WindbondSPIFlash". I will open an issue at his repo.

per1234, 2018-02-10:
It should be noted that the issue is only related to the value of the `name` property used by the Arduino IDE's Library Manager, not to the name of the repository or the source filenames. The `name` property is only used for:

- The list of installed libraries at **Sketch > Include Library > Manage Libraries**.
- The name of the library in the File > Examples menu.
- The name of the library in the Sketch > Include Library menu.
- Checking against the libraries in the Library Manager index to see if a newer version exists (thus the spurious update notifications).

Since your library doesn't have a library.properties file, the Arduino IDE automatically sets the value of the `name` property to the library folder name. In fact the library is probably named "SPIFlash-master" for many users since the **Sketch > Include Library > Add .ZIP Library** installation method is so popular and the branch name is appended to the folder name of the file you get from GitHub's **Clone or download > Download ZIP**.

The simple workaround would be to add a file named library.properties to the root of the repository that sets the name property to a unique value. For example:
```
name=SPIFlash_LowPowerLab
```
This really has no impact on the users of your library other than preventing them from getting spurious update notifications.

The other workaround would be to set the `version` property in your library.properties to something significantly higher than the other SPIFlash library, causing the installed version of this library to always appear newer than the ones available in the Library Manager index. For example:
```
version=101.0.0
```
That would prevent spurious update notifications while allowing you to keep the `name` property set to SPIFlash.

library.properties format specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

LowPowerLab, 2018-02-12:
@per1234 Thank you for the suggestion, how about the **library.json** file?
That already has a `name=SPIFlash` property. Should that be renamed to match `name=SPIFlash_LowPowerLab` in **library.properties**?

per1234, 2018-02-12:
The library.json file is not used by the Arduino IDE. It's used by platformio, which I don't have any experience with. I see your library is already in the platformio library manager so it might be problematic to change the name. I know that with the Arduino Library Manager if you want to change the library name you need to submit a request on the arduino/Arduino issue tracker for the index to be updated. After the name change the people who had the library installed with the previous name will not get update notifications from Arduino Library Manager when a new version is released (which is why Marzogh has been hesitant to chance the name of their library). However, I don't know whether any of that applies to platformio's library manager system, which was developed independently from Arduino's.

LowPowerLab, 2018-02-14:
@per1234 thank you again, here is the arduino issue for reference:
https://github.com/arduino/Arduino/issues/7217

per1234, 2018-02-14:
The options I suggested earlier were meant as an either/or. It's not a problem that you did both, I just wanted to be clear about what I meant.

Do I understand correctly that arduino/Arduino#7217 is a request for your library to be added to the Library Manager index? If so then the `name` property change was the necessary thing to do since each library in the Library Manager index is required to have a unique name so really I only provided one viable option.

LowPowerLab, 2018-02-14:
@per1234 Yes, it's a request to update it, or add it in case it's not there.
I would hope the SPIFlash_LowPowerLab makes it unique.

per1234, 2018-02-14:
Your library has never been in the Library Manager index. The Marzogh/SPIFlash is in the Library Manager index and this caused people with your library installed to get spurious update notifications since Library Manager sees that:
1. A library of the name SPIFlash (assigned according to the folder name since this library didn't previously have a library.properties file) is installed.
2. The version of the installed library (undefined since this library didn't previously have a library.properties file) was lower than the version available in the Library Manager index.

Actually I see that at one time someone did submit a request for this library to be added to the Library Manager index (https://github.com/arduino/Arduino/issues/5304) but it was never able to be done since the requirements were not met. It's not clear whether the person who submitted that request was actually authorized by LowPowerLab to do so.

LowPowerLab, 2018-02-14:
@per1234 I was not aware about that other request. I guess I am really behind with the rules and how these things work. I learned how to add libraries way back when you just copied them in your `arduino/libraries` folder.
So ... as things stand now, is my request to add this library looking OK?
Will this solve my problem of separation from the other `Marzogh/SPIFlash` conflicting name library?

per1234, 2018-02-14:
>I learned how to add libraries way back when you just copied them in your arduino/libraries folder.

That's still definitely valid but Library Manager is a little more beginner friendly. Also, the update notifications feature of Library Manager is really cool (except when it's causing the problem originally described in this issue report). The Library Manager indexer will pick up any new release you make within an hour or so and notify all the people who have installed the library that a new version is available. They can also easily roll back to previous versions, which is helpful for troubleshooting.

>is my request to add this library looking OK?

See my comment: https://github.com/arduino/Arduino/issues/7217#issuecomment-365618723. Once that's resolved your library will meet all the requirements. Then you only need to wait for the issue to be closed by the Arduino developer in charge of adding submitted libraries to the index. Shortly after that the library will be available for installation via Library Manager!

>Will this solve my problem of separation from the other Marzogh/SPIFlash conflicting name library?

Adding your library to the Library Manager index is not necessary to solve that problem and doesn't do anything to help or harm the problem. Either changing the name property or setting the version higher than the version of Marzogh/SPIFlash alone solves that problem. So the Library Manager addition is simply a way of making your library easier for the users to install and update.

LowPowerLab, 2018-02-20:
It is [done](https://github.com/arduino/Arduino/issues/7217#event-1481977555).
Closing this issue.

closed 2018-02-20 by LowPowerLab