LowPowerLab/SPIFlash #20: using SPIFlash with arduino NANO

papilinou, 2018-05-22:
SPI.begin must be call even in the SPI library with " transaction" included in the Arduino IDE 1.8.5.
See that example : https://www.arduino.cc/en/Tutorial/SPITransaction.
A good place to include it could be in SPIFlash initialize around line 94 and delete around line 70
/// setup SPI, read device ID etc...
boolean SPIFlash::initialize()
{
  _SPCR = SPCR;
  _SPSR = SPSR;
  pinMode(_slaveSelectPin, OUTPUT);
  SPI.begin();
#ifdef SPI_HAS_TRANSACTION
  _settings = SPISettings(4000000, MSBFIRST, SPI_MODE0);
#endif

also to get the complete jedecID (3 bytes):
uint32_t SPIFlash::readDeviceId()
{
#if defined(__AVR_ATmega32U4__) // Arduino Leonardo, MoteinoLeo
  command(SPIFLASH_IDREAD); // Read JEDEC ID
#else
  select();
  SPI.transfer(SPIFLASH_IDREAD);
#endif
  
uint32_t jedecid = SPI.transfer(0)  ;
jedecid = jedecid << 16 & 0x00ffffff ;
  jedecid |= SPI.transfer(0) << 8 ;
  jedecid |= SPI.transfer(0);
  unselect();
  return jedecid;
}

LowPowerLab, 2018-05-23:
Thank you for the suggestion!

I moved SPI.begin() in [this commit](https://github.com/LowPowerLab/SPIFlash/commit/eec4aa34b26242fa8ff6b2bf6f084d721c158044).

For the JEDEC ID, maybe this can come later. I will leave this issue OPEN for this reason.

harissutanrafiq, 2018-09-29:
its possible write and read float value ..?  can you give me tutorial..?

LowPowerLab, 2018-10-01:
@harissutanrafiq You can write anything, in byte format.