LowPowerLab/SPIFlash #22: 'SS_FLASHMEM' was not declared in this scope

mitch85, 2019-08-11:
Hi, i use the node.ino example of the RFM69 Library. But on compiling i got these error the obove error....

LowPowerLab, 2019-08-12:
You have to install and select a Moteino board, or replace SS_FLASHMEM with your SPI CS for that chip.

closed 2019-08-12 by LowPowerLab

Erwin-66, 2020-06-12:
What would be the SPI CS for the Arduino Uno. I still get the same error

LowPowerLab, 2020-06-12:
@Erwin-66 Use constructor to pass in your board's SPI CS pin.

`SPIFlash(uint8_t slaveSelectPin, uint16_t jedecID=0);`

So call `SPIFlash(CS)` or `SPIFlash(10)` for specific pin if `CS` is not defined.

LowPowerLab, 2023-04-13:
You need to declare the SPI CS (SS_FLASHMEM) pin where this chip is connected, if you don't have the [Moteino boards installed](https://lowpowerlab.com/guide/moteino/programming-libraries/).
Ex with D10 as SPI CS for the flash chip:
#declare SS_FLASHMEM 10

datajournalist12, 2023-11-05:
Note for anyone following this issue that the following worked for me for Arduino Nano:

#define SS_FLASHMEM 10
SPIFlash flash(SS_FLASHMEM);