LowPowerLab/SPIFlash #26: Can hang in busy() if Flash chip is powered down and there is noise on MISO

JasonC0x0D, 2021-06-04:
1. Comment in code about using a pull down resistor can work, but other SPI devices like SD cards may need a pull up resistor.
2. Better solution seems to be to be to track if the chip is powered down or not and if it is powered down, not to read any data on the MISO line since it will just be noise.
3. Looked into other another SPI library and found they had this feature built in using a variable they called chipPoweredDown
- https://www.arduino.cc/reference/en/libraries/spimemory/ 
- https://github.com/Marzogh/SPIMemory

Note: Bug found when using:
**Hardware**
Moteino Trace Antenna RFM69HCW 915MHz with Flash Memory Chip
LowPowerLab BME280 Board
1 400mAh LiPo battery (from Adafruit)
1 MOhm resistor from VIN to A7 - Battery Monitor
1 MOhm resistor from A7 to Ground
1 Solderless breadboard

**Software**
Default LowPowerLab WeatherMote Example sketch that was configured to the network.

LowPowerLab, 2021-06-07:
Sorry I haven't had a chance to reply properly yet. I believe the issue is that after a flash.sleep() command, only the device_id and wake()  commands will be responsive, see section 8.2.19 from the w25x40CL datasheet. This is **normal behavior** :-)
I chose not to pollute the library with variables that keep track of what the user is doing. I believe that is something that the user should do in their firmware code. If you issue a sleep(), then first thing upon wakeup is to call flash.wake() to use it. Otherwise it is unresponsive, as per the datasheet.

JasonC0x0D, 2021-06-07:
Felix,

I agree that the flash chip is responding as stated in the data sheet. 

1. I can see both pros and cons with leaving it up to the user to properly use the sleep and wake commands. At the end of the day that is a preference and I can defer to the original author. 
2. I would then request a change to the comment in the library recommending a pull down resistor on the MISO. I have read but not tested that some SD cards need pull up resistors on MISO. I would request the comment would state when the flash chip is sleeping (powered down) only two commands are valid, wake up or request id.
3. This means the weather mote example sketch will need to be updated as it calls flash.sleep() without regard to the state of the flash chip. There are possibly other examples that may need this update.

- Thanks, Jason

LowPowerLab, 2021-06-07:
1. OK.
2. A resistor in case it's needed, on a SPI MISO line, is a no go unfortunately. Never seen a need for one as long as the SPI device is used appropriately.
3. I agree, I will plan to add more comments and update examples that call sleep() multiple times without calling wake(). Ultimately the user needs to keep track or code for the spiflash to be asserted correctly and not make assumptions. The library has no business doing that, especially considering that upon a MCU soft reset a spiflash chip can stay in a sleep() condition.

commit 9031099c (2021-06-07): Update SPIFlash.cpp

Closes #26, Closes #27
closed 2021-06-07 by LowPowerLab